AUCOHL/DFFRAM #11: The table in Handcrafted file is hard to understand

mithro, 2021-04-11:
The table in https://github.com/Cloud-V/DFFRAM/tree/main/Handcrafted#hardening-the-memory is hard to understand.

I was expecting to see a table which has the DFFRAM and automatic PnR results side by side.

IE;
 * 1kb ram in automatic PnR is 60% density, with DFFRAM it is 85% density.
 * 1kb ram in automatic PnR is XXXum2, with DFFRAM it is YYYum2.

mithro, 2021-04-11:
The table in your top level README is much better - https://github.com/Cloud-V/DFFRAM#handcrafted

donn, 2021-04-11:
Yeah, was about to say. Could probably just transplant this table into the Handcrafted Readme as well, but that's Prof. Shalan territory. @shalan Thoughts?

commit 52e78763 (2021-04-13): Fix #11
closed 2021-04-13 by donn