AUCOHL/DFFRAM #94: GDS Output?

dan-fritchman, 2021-06-29:
I see that the DFFRAM Overview: 
https://github.com/Cloud-V/DFFRAM#overview

Says:
```
Different views (HDL netlist, HDL functional models, LEF, GDS, Timing, …) are all generated for a given size configuration.
```

After running the compiler, I don't see any GDS. 

Is that the idea?  
Is there some follow-up step which should be generating one?  
Or is this an indication that something went wrong?

donn, 2021-06-29:
We honestly just haven't gotten around to doing the GDS step yet. Caught up in some issues. :(

dan-fritchman, 2021-06-30:
So, how do (or would) _you_ go about producing a GDS?  
Load up one of the cells in Magic and export it?

Manarabdelaty, 2021-06-30:
@dan-fritchman Yes, you can use magic to generate the gds. I would follow the commands here to generate the gds file from the def file https://github.com/efabless/OpenLane/blob/master/scripts/magic/mag_gds.tcl , `TECH_LEF` should point to the technology lef under your PDK_ROOT (`libs.ref/sky130_fd_sc_hd/techlef/sky130_fd_sc_hd.tlef`) and assume that `MAGIC_PAD` and `MAGIC_ZEROIZE_ORIGIN` are set to zero

donn, 2021-07-28:
GDS output has been added, closing.

closed 2021-07-28 by donn