AUCOHL/DFFRAM #96: String error in dffram.py

apond308, 2021-07-07:
When running dffram.py, getting TypeError with string path for pdk_root:

File "./dffram.py", line 692, in flow
    prep(build_folder, pdk_root)
  File "./dffram.py", line 117, in prep
    pdk_root = os.path.abspath(os.path.realpath(local_pdk_root))
  File "/usr/lib/python3.8/posixpath.py", line 390, in realpath
    filename = os.fspath(filename)
TypeError: expected str, bytes or os.PathLike object, not NoneType

apond308, 2021-07-07:
How do I open a branch to make the change?

donn, 2021-07-08:
Hey apond, generally, when you want to make a PR to an organization you're not a member of, you start by forking the relevant repository (e.g. `apond308/DFFRAM`,) make the changes, then create "Open a Pull Request" there.

apond308, 2021-07-08:
Great, thank you. I will go ahead and do that.

apond308, 2021-09-15:
Issue addressed in pull request #106

closed 2021-09-15 by apond308