AUCOHL/DFFRAM #110: Global routing hits "Invalid 2D tree" error

antonblanchard, 2021-10-30:
I'm still trying to build native 64 bit RAMs, but in the meantime I thought I'd try doubling up 32 bit RAMs. I get the following error when building a 1024x32 RAM:

```
dffram.py -s 1024x32
...
[INFO GRT-0101] Running extra iterations to remove overflow.
[INFO GRT-0103] Extra Run for hard benchmark.
[WARNING GRT-0164] Initial grid wrong y1 x1 [114 13], net start [114 14] routelen 15.
[ERROR GRT-0167] Invalid 2D tree for net BANK512\[0\].RAM512.BANK128\[0\].RAM128.Do0\[18\].
Error: route.tcl, 21 GRT-0167
```

donn, 2021-10-31:
Can you get the .placed.def  file so I can forward it to the OpenROAD team?

antonblanchard, 2021-10-31:
Sure, here it is: [RAM1024.placed.def.gz](https://github.com/Cloud-V/DFFRAM/files/7449633/RAM1024.placed.def.gz)

donn, 2021-11-02:
Hmm, this appears to be fixed in newer versions of OpenROAD... Let me see if an OpenLane upgrade fixes this

donn, 2021-11-02:
Nah, we're on the latest OpenLane release. Which means I have to release a new version of OpenLane with a new version of OpenROAD...

donn, 2022-01-20:
Should be okay now.

closed 2022-01-20 by donn