AUCOHL/DFFRAM #115: dffram.py fails at step STA

andrsmllr, 2021-11-14:
Hi,

I'm having troubles generating any DFFRAM from the set of supported parameters, e.g. 128x32.
I tried the DFFRAM main branch and also checked out tag 0.0.1, getting seemingly the same result.

The sky130A PDK (latest) is installed under:
```
$>echo $PDK_ROOT
/opt/open_pdks/share/pdk
```
The PDK appears to be recognized, since `./dffram.py -s 128x32` starts running through synthesis etc., but then fails at static timing analysis.
On the first run of `dffram.py` the docker image `efabless/openlane:2021.10.25_20.35.00` was pulled. There is also a docker image named `dffram-env:latest`, but I didn't catch where it came from...

Here are the steps I have taken:
```
$>git clone https://github.com/Cloud-V/DFFRAM
$>cd DFFRAM
$>./dffram -s 128x32
OpenROAD GITDIR-NOTFOUND This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
mergeLef.py : Merging LEFs
sky130_fd_sc_hd.lef: SITEs matched found: 0
sky130_fd_sc_hd.lef: MACROs matched found: 441
mergeLef.py : Merging LEFs complete
--- Synthesis ---

[... more output here ...]

     sky130_fd_sc_hd__mux4_1        32
     sky130_fd_sc_hd__nor3b_2        5
     sky130_fd_sc_hd__nor4b_2       16

   Chip area for module '\RAM128': 138625.452800

--- Static Timing Analysis ---
OpenROAD GITDIR-NOTFOUND 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO ODB-0222] Reading LEF file: ./build/128x32_DEFAULT/merged.lef
[INFO ODB-0223]     Created 13 technology layers
[INFO ODB-0224]     Created 25 technology vias
[INFO ODB-0225]     Created 441 library cells
[INFO ODB-0226] Finished LEF file:  ./build/128x32_DEFAULT/merged.lef
A step has failed: Command '['docker', 'run', '--rm', '-v', '/opt/open_pdks/share/pdk:/opt/open_pdks/share/pdk', '-v', '/home/user/git/DFFRAM:/mnt/dffram', '-w', '/mnt/dffram', '-e', 'PDK_ROOT=/opt/open_pdks/share/pdk', '-e', 'PDKPATH=/opt/open_pdks/share/pdk/sky130A', '-e', 'LC_ALL=en_US.UTF-8', '-e', 'LANG=en_US.UTF-8', 'efabless/openlane:2021.10.25_20.35.00', 'openroad', '-exit', './build/128x32_DEFAULT/sta.tcl']' returned non-zero exit status 139.
Quick invoke: docker run --rm -v /opt/open_pdks/share/pdk:/opt/open_pdks/share/pdk -v /home/user/git/DFFRAM:/mnt/dffram -w /mnt/dffram -e PDK_ROOT=/opt/open_pdks/share/pdk -e PDKPATH=/opt/open_pdks/share/pdk/sky130A -e LC_ALL=en_US.UTF-8 -e LANG=en_US.UTF-8 efabless/openlane:2021.10.25_20.35.00 openroad -exit ./build/128x32_DEFAULT/sta.tcl
```

Have I overlooked some dependency or parameter? I also don't quite understand why it says `OpenROAD GITDIR-NOTFOUND`; shouldn't everythin be provided by the docker image?
My system is Ubuntu 20.04 running in WSL2 on a Windows 10 host.

donn, 2021-11-14:
Hey- so for background: `dffram-env:latest` is an ad hoc image that was built and used by some older versions of dffram, but is no longer needed.

This STA failure looks like a memory error of some kind: if you're using the latest PDK, that'd be the cause, as we do have a confirmed issue with that particular PDK: see https://github.com/The-OpenROAD-Project/OpenROAD/issues/1305 for more info. We'll ping you when there's an available fix in OpenLane.

donn, 2022-02-04:
Staled out.

closed 2022-02-04 by donn