AUCOHL/DFFRAM #119: dffram should allow pointing to a local openroad install

proppy, 2021-12-13:
https://github.com/Cloud-V/DFFRAM/blob/main/docs/Usage.md#using-dffram seems to suggest `docker` is only required to run `placeram` but in practice almost all the steps depends on the `openlane` function `run_docker` invocation: https://github.com/Cloud-V/DFFRAM/blob/main/dffram.py#L77

It would be nice to optionally allow users to point to a local openroad install in order to allow them to run it in environment where docker is not supported (i.e: https://colab.research.google.com/).

donn, 2021-12-13:
It's a little bit less straightforward than that since we also do rely on a number of scripts inside the OpenLane container and we also rely on some OpenLane utilities such as magic for sign-off.

That said, assuming someone is willing to download an OpenLane repo and point to it + compile magic themselves, I'm not opposed to the idea.

donn, 2021-12-13:
Yeah, that's a lot more doable. I'll look into it.

proppy, 2022-03-23:
Looks like this was fixed with #143?