AUCOHL/DFFRAM #130: Add demos of using the tool in Juypter notebooks / Google Colabs

mithro, 2022-02-03:
It would be great if DFFRAM examples and documentation was using Juypter Notebooks / Google Colabs to allow people to play with things extremely quickly.

@proppy has done a bunch of work to show this is totally possible;
 - https://twitter.com/proppy/status/1482036203428147203?s=20
 - https://twitter.com/proppy/status/1472768213381697540?s=20

You can then even add a "Launch in Colab" button / badge in the README...

proppy, 2022-02-09:
one of the pre-requisite to this, is to get openlane and its dependencies to work well in https://colab.research.google.com/ which is not "trivial" in the current state of things because container can't really work on colab: ttps://github.com/googlecolab/colabtools/issues/299.

this should be easier as we get more things packaged in conda:
https://github.com/hdl/conda-eda/pull/164
https://github.com/hdl/conda-eda/pull/165

donn, 2022-03-21:
Done

closed 2022-03-21 by donn

proppy, 2022-03-23:
Didn't realize this was being fixed with #143, added a few comments here https://github.com/Cloud-V/DFFRAM/pull/143#pullrequestreview-918290992 but let m me know if you'd like me to send follow-up PRs to address them.