AUCOHL/DFFRAM #148: Very large max transition violations

antonblanchard, 2022-04-03:
I'm seeing some huge max transition violations when building various DFFRAMs. An example using `32x64_1RW1R`:

```
Pin                                    Limit    Slew   Slack
------------------------------------------------------------
SLICE[0].RAM8.DEC1.AND0/Y               1.50    6.48   -4.98 (VIOLATED)
SLICE[0].RAM8.WORD[0].W.SEL1BUF/A       1.50    6.48   -4.98 (VIOLATED)
SLICE[3].RAM8.DEC1.AND0/Y               1.50    5.89   -4.39 (VIOLATED)
SLICE[3].RAM8.WORD[0].W.SEL1BUF/A       1.50    5.89   -4.39 (VIOLATED)
SLICE[1].RAM8.WORD[0].W.SEL1BUF/A       1.50    5.70   -4.20 (VIOLATED)
SLICE[1].RAM8.DEC1.AND0/Y               1.50    5.70   -4.20 (VIOLATED)
SLICE[2].RAM8.DEC1.AND0/Y               1.50    5.43   -3.93 (VIOLATED)
SLICE[2].RAM8.WORD[0].W.SEL1BUF/A       1.50    5.43   -3.93 (VIOLATED)
SLICE[0].RAM8.WORD[7].W.BYTE[0].B.SEL0INV/A    1.50    1.91   -0.41 (VIOLATED)
SLICE[0].RAM8.WORD[7].W.BYTE[0].B.CGAND/A    1.50    1.91   -0.41 (VIOLATED)
SLICE[0].RAM8.WORD[7].W.BYTE[1].B.SEL0INV/A    1.50    1.90   -0.40 (VIOLATED)
SLICE[0].RAM8.WORD[7].W.BYTE[1].B.CGAND/A    1.50    1.90   -0.40 (VIOLATED)
SLICE[0].RAM8.WORD[7].W.BYTE[2].B.CGAND/A    1.50    1.90   -0.40 (VIOLATED)
SLICE[0].RAM8.WORD[7].W.BYTE[2].B.SEL0INV/A    1.50    1.90   -0.40 (VIOLATED)
SLICE[0].RAM8.WORD[7].W.BYTE[3].B.CGAND/A    1.50    1.90   -0.40 (VIOLATED)
SLICE[0].RAM8.WORD[7].W.BYTE[3].B.SEL0INV/A    1.50    1.90   -0.40 (VIOLATED)
```

The worst of it appears to be this path:

```
Fanout     Cap    Slew   Delay    Time   Description
-----------------------------------------------------------------------------
                          0.00    0.00   clock CLK (rise edge)
                          0.00    0.00   clock network delay (propagated)
                          3.62    3.62 ^ input external delay
                  0.08    0.06    3.68 ^ EN1 (in)
     9    0.05                           EN1 (net)
                  0.08    0.00    3.68 ^ EN1BUF/A (sky130_fd_sc_hd__clkbuf_2)
                  0.07    0.15    3.83 ^ EN1BUF/X (sky130_fd_sc_hd__clkbuf_2)
     4    0.01                           DEC1.EN (net)
                  0.07    0.00    3.83 ^ DEC1.AND0/C_N (sky130_fd_sc_hd__nor3b_2)
                  0.15    0.19    4.02 ^ DEC1.AND0/Y (sky130_fd_sc_hd__nor3b_2)
     1    0.01                           SLICE[0].RAM8.DEC1.EN (net)
                  0.15    0.00    4.02 ^ SLICE[0].RAM8.DEC1.ENBUF/A (sky130_fd_sc_hd__clkbuf_2)
                  0.11    0.20    4.22 ^ SLICE[0].RAM8.DEC1.ENBUF/X (sky130_fd_sc_hd__clkbuf_2)
     8    0.02                           SLICE[0].RAM8.DEC1.EN_buf (net)
                  0.11    0.00    4.22 ^ SLICE[0].RAM8.DEC1.AND0/D_N (sky130_fd_sc_hd__nor4b_2)
                  6.48    5.01    9.23 ^ SLICE[0].RAM8.DEC1.AND0/Y (sky130_fd_sc_hd__nor4b_2)    <---- HERE I AM
     1    0.30                           SLICE[0].RAM8.WORD[0].W.SEL1 (net)
                  6.48    0.11    9.34 ^ SLICE[0].RAM8.WORD[0].W.SEL1BUF/A (sky130_fd_sc_hd__clkbuf_2)
                  1.52    1.73   11.08 ^ SLICE[0].RAM8.WORD[0].W.SEL1BUF/X (sky130_fd_sc_hd__clkbuf_2)
     8    0.29                           SLICE[0].RAM8.WORD[0].W.BYTE[0].B.SEL1 (net)
                  1.52    0.09   11.16 ^ SLICE[0].RAM8.WORD[0].W.BYTE[2].B.SEL1INV/A (sky130_fd_sc_hd__inv_1)
                  0.51    0.64   11.80 v SLICE[0].RAM8.WORD[0].W.BYTE[2].B.SEL1INV/Y (sky130_fd_sc_hd__inv_1)
     8    0.05                           SLICE[0].RAM8.WORD[0].W.BYTE[2].B.SEL1_B (net)
                  0.51    0.01   11.81 v SLICE[0].RAM8.WORD[0].W.BYTE[2].B.BIT[1].OBUF1/TE_B (sky130_fd_sc_hd__ebufn_2)
                  0.00    0.23   12.04 v SLICE[0].RAM8.WORD[0].W.BYTE[2].B.BIT[1].OBUF1/Z (sky130_fd_sc_hd__ebufn_2)
     2    0.24                           Do1_REG.Di[17] (net)
                  0.00    0.10   12.14 v Do1_REG.OUTREG_BYTE[2].Do_FF[1]/D (sky130_fd_sc_hd__dfxtp_1)
                                 12.14   data arrival time
```

A few things stand out:

- Why are we using clkbuf instances? We shouldn't need them, and the higher capacitance compared to a regular buf isn't helping.
- `sky130_fd_sc_hd__nor4b_2` is only characterised out to 0.065 pF and we are driving it with 0.300 pF of capacitance. It's not clear to me why there is so much capacitance on this net (long wire?) but we are operating way outside of the liberty timing table and it makes me worried our STA results could be bad.

donn, 2022-04-04:
@shalan

donn, 2022-04-04:
Didn't mean to close that, apologies.

antonblanchard, 2022-04-07:
@shalan any suggestions here? The Microwatt MPW5 tape out used this DFFRAM and I was wondering if the STA timing results are going to be suspect.

antonblanchard, 2022-05-25:
I think some of these problems were a result of the suspect RCX data. Things look better with the latest tools.

Also, having worked through a number of issues while integrating DFFRAM into Microwatt, I think we need to produce liberty files. They don't have to be perfect, just provide (worst case) setup and hold constraints to OpenROAD when integrating DFFRAM into other modules.