AUCOHL/DFFRAM #156: Is the colab broken?

tcal-x, 2022-05-02:
In the first code cell output I see a "restart" -- is that expected?

⏬ Downloading https://repo.anaconda.com/miniconda/Miniconda3-py37_4.11.0-Linux-x86_64.sh...
📦 Installing...
📌 Adjusting configuration...
🩹 Patching environment...
⏲ Done in 0:00:19
🔁 Restarting kernel...

Then in the second code cell,
```
---------------------------------------------------------------------------
AssertionError                            Traceback (most recent call last)
[<ipython-input-2-593fad972db4>](https://localhost:8080/#) in <module>()
      1 import condacolab
      2 
----> 3 condacolab.check()
      4 get_ipython().system('sed -i -e /cudatoolkit/d /usr/local/conda-meta/pinned')

[/usr/local/lib/python3.7/dist-packages/condacolab.py](https://localhost:8080/#) in check(prefix, verbose)
    302     assert (
    303         f"{prefix}/lib" in os.environ["LD_LIBRARY_PATH"]
--> 304     ), f"💥💔💥 LD_LIBRARY_PATH was not patched! Value: {os.environ['LD_LIBRARY_PATH']}"
    305     if verbose:
    306         print("✨🍰✨ Everything looks OK!")

AssertionError: 💥💔💥 LD_LIBRARY_PATH was not patched! Value: /usr/local/nvidia/lib:/usr/local/nvidia/lib64
```

proppy, 2022-05-03:
That's a limitation of https://github.com/conda-incubator/condacolab you have to leave some time for the runtime to boot after the first restart.

You can check that everything is in place by running:
```
import condacolab
condacolab.check()
```

tcal-x, 2022-05-03:
I had tried forging ahead before, and ran into later issues, so I assumed that the first error I saw was the root cause.   Now after waiting between the first and second cells, I still get this -- it seems easily fixable, but is it intentional?

[image]

proppy, 2022-05-03:
Seems like the notebook miss a few dependencies, in order for this to succeed:
```
    import click
    import yaml
    import volare
```
Looks like openlane doesn't have `volare` in its [ `requirements.txt`](https://github.com/The-OpenROAD-Project/OpenLane/blob/master/requirements.txt) so that should either be added there or in the notebook itself.

proppy, 2022-05-03:
Note that `volare` itself is not a strict requirement for `DFFRAM` (since we install `open_pdks.sky130` thru conda) but the main script currently expect it to be installed in `PYTHONPATH` (even if it's only used if the pdk is not provided)

proppy, 2022-05-03:
alternatively the following patch should also workaround the issue:
```
diff --git a/dffram.py b/dffram.py
index 499c248..a528a9c 100755
--- a/dffram.py
+++ b/dffram.py
@@ -1,5 +1,5 @@
 #!/usr/bin/env python3
-# -*- coding: utf8 -*-
+# -*- coding: utf-8 -*-
 # Copyright ©2020-2021 The American University in Cairo and the Cloud V Project.
 #
 # This file is part of the DFFRAM Memory Compiler.
@@ -21,7 +21,6 @@ import os
 try:
     import click
     import yaml
-    import volare
 except ImportError:
     print(
         "You need to install dependencies: pip3 install --user --upgrade --no-cache-dir -r ./requirements.txt"
@@ -142,6 +141,7 @@ def prep(local_pdk_root):
     pdk_root = os.path.abspath(local_pdk_root)
     pdk_path = os.path.join(pdk_root, pdk)
     if not os.path.exists(os.path.join(pdk_path)):
+        import volare
         print(f"PDK not found at {pdk_path}, grabbing PDK using volare...")
         volare.enable(pdk_root=local_pdk_root, pdk=pdk_family, version=pdk_version)
```

tcal-x, 2022-11-11:
I gave this another try with the current dffram.ipynb, and encountered this:
```
[INFO ODB-0134] Finished DEF file: ./build/32x32_DEFAULT/RAM32.placed.def
Placement successful.
--- Hardening With OpenLane ---
OpenLane UNKNOWN
(with mounted scripts from fatal: not a git repository: '/content/openlane/.git')
All rights reserved. (c) 2020-2022 Efabless Corporation and contributors.
Available under the Apache License, version 2.0. See the LICENSE file for more details.

[ERROR]: Container manifest not found. What this likely means is that the container is severely out of date.
child process exited abnormally
[ERROR]: Please update your environment. OpenLane will now quit.
child process exited abnormally
    while executing
"exec tclsh {*}$args >&@stdout"
    (procedure "run_file" line 3)
    invoked from within
"run_file [file normalize $arg_values(-file)] {*}$argv"
    invoked from within
"if { [info exists flags_map(-interactive)] || [info exists flags_map(-it)] } {
    if { [info exists arg_values(-file)] } {
        run_file [file nor..."
    (file "/content/openlane/flow.tcl" line 401)
A step has failed: Command '['flow.tcl', '-design', './build/32x32_DEFAULT/openlane', '-it', '-file', './build/32x32_DEFAULT/openlane/interactive.tcl']' returned non-zero exit status 1.
Quick invoke: flow.tcl -design ./build/32x32_DEFAULT/openlane -it -file ./build/32x32_DEFAULT/openlane/interactive.tcl
```