AUCOHL/DFFRAM #189: "nix-shell" command's execution time

Shehab-Naga, 2023-09-21:
Greetings,

I am in the process of installing and using the DFFRAM compiler, and I am following the steps mentioned in "Usage.md" file (https://github.com/AUCOHL/DFFRAM/blob/main/docs/Usage.md). However, since I ran the "nix-shell" command, it is taking so much time, almost 3 hours now. I am writing to ask if it is normal for the "nix-shell" to be running for a long time or there is a mistake somewhere.

As a side note, maybe it helps in identifying the problem, I am using openLane from this repo (https://github.com/The-OpenROAD-Project/OpenLane).

Thanks in advance.

donn, 2023-09-22:
This may happen if you forgot the Cachix step or did not execute it correctly- it will build every tool included with OpenLane from scratch. Needless to say, this will take forever: https://github.com/AUCOHL/DFFRAM/blob/main/docs/Usage.md#setting-up-the-binary-cache

Some logs would be helpful.

> As a side note, maybe it helps in identifying the problem, I am using openLane from this repo ([The-OpenROAD-Project/OpenLane](https://github.com/The-OpenROAD-Project/OpenLane)).

DFFRAM pulls its own OpenLane as part of `nix-shell`, so that's not an issue.

Shehab-Naga, 2023-09-23:
Thanks for your answer. 

I did the Cachix step correctly. Here is what happened. It took this long time only the first time I executed this command, but now it opens immediately.

 Another question, is there an instructions file for porting the DFFRAM compiler to another PDK other than "gf180mcu_fd_sc_mcu7t5v0" and "sky130_fd_sc_hd"?

donn, 2023-09-27:
> It took this long time only the first time I executed this command, but now it opens immediately.

So as a small primer, nix will cache the result of any `nix-shell` on your computer, which is why it doesn't take that long the second time.

What Cachix tries to do is insert an alternative cache where OpenLane's dependencies are already built, so you don't have to do that too. So it's weird that it didn't pull dependencies.

THAT SAID; out of curiosity, are you on an Apple Silicon Mac? If so, that would explain the lack of binaries.

closed 2023-12-18 by donn