AUCOHL/DFFRAM #191: unknown compression method 'zstd'

b224hisl, 2023-12-17:
Hello, I'm installing DFFRAM using the nix docker. (https://nixos.org/download.html#nix-install-docker)
And I followed the steps in https://github.com/AUCOHL/DFFRAM/blob/main/docs/Usage.md. 
But I met a problem when executing `nix-shell`, it reports `unknown compression method 'zstd'`, shown as:
[image]
I have highlighted my commands. So does anybody knowns how to handle this problems?

b224hisl, 2023-12-17:
Actually, I have installed the zstd
[image]

donn, 2023-12-17:
Not really sure why this would happen. Are you sure you installed Nix using `sh <(curl -L https://nixos.org/nix/install) --daemon --yes` or did you `apk add nix`?

b224hisl, 2023-12-18:
> Not really sure why this would happen. Are you sure you installed Nix using `sh <(curl -L https://nixos.org/nix/install) --daemon --yes` or did you `apk add nix`?

No because I use the nix docker so I think it should have nix

donn, 2023-12-18:
The Nix docker is not up-to-date. I'll kindly ask you to either install Nix natively.

b224hisl, 2023-12-19:
> The Nix docker is not up-to-date. I'll kindly ask you to install Nix natively.

Well maybe you are right. After giving up on using Nixdocker, I reinstalled Nix and that resolved the issue. Anyway thank you very much~ Now you can close this issue.

donn, 2023-12-19:
Glad it worked!

closed 2023-12-19 by donn