AUCOHL/DFFRAM #192: Support for 4R2W and up to 9000

Dolu1990, 2024-01-06:
Hi,

I have a RISC-V dual issue core (need 4R2W 32 bits 64 words) and was looking for a register file implementation.
I first tried to use latch + tristate based HDL directly on openlane / sky130, it goes into fatal congestion error even at 20% density XD

This issue is mostly to notify the interrest for the tool. I'm not expecting anybody to implement this feature.

Thanks anyway :)

donn, 2024-01-07:
Up to 9000 what, if you don't mind clarifying?

Dolu1990, 2024-01-07:
9000 is for the meme : 
https://knowyourmeme.com/memes/its-over-9000

But overall the intends is to go beyond 2R1W configs

donn, 2024-01-07:
The meme is "over 9000." ;)

Not sure of the viability of more ports simply by adding more multiplexers; as each port adds a ton more congestion as you've figured out trying to do it with OpenLane. The resource @mithro shared is interesting however: https://tomverbeure.github.io/2019/08/03/Multiport-Memories.html

Dolu1990, 2024-01-16:
@donn 

> the resource @mithro shared is interesting however: https://tomverbeure.github.io/2019/08/03/Multiport-Memories.html

Thanks :)
That is what i use on FPGA
Thing is, to translate a 2R1W into a 4R2W, you need 4x 2R1W, which hurts my ASIC feelings ^^ (ok on FPGA, as there is no alternative)