AUCOHL/DFFRAM #198: 1RW1R support

maliberty, 2025-07-04:
The message `1RW1R variants are temporarily disabled due to a bug.` is two years old.  Is there a plan to resolve this?  I'd like to regenerate the RAMs for microwatt but they require this case that used to be supported.

donn, 2025-07-04:
As I recall there's a pretty bad timing issue with it. We don't have anyone working on DFFRAM right now so I wouldn't count on it.

If your goal is recreating an older design, it may be your best bet to use a version of DFFRAM from the time...

donn, 2025-07-04:
I… yeah 1RW1R appears to be routing fine… [image]

Static timing's alright at a 20ns clock:

[image]

Still weary of recommending you use them for production because obviously 2022 me must have have had a good reason but if it's to recreate an old design, should be fine. I will however add a warning to dffram.py.